AprilRobotics/apriltag_ros #1: Question about how to build and use this package

ChunJyeBehBeh, 2018-02-14:
(i) cd catkin_ws/src/ 
(ii) git clone https://github.com/dmalyuta/apriltags2_ros.git
(iii) cd catkin_ws/ & catkin_make
Is that all the steps required to build the package? Anything I miss out?

Run the package
Terminal 1: roscore
Terminal 2: 
Terminal 3: 
Terminal 4: 
Any other commands? I m not sure about that and is a newbie to ROS

dmalyuta, 2018-02-14:
That is correct, on my system (Ubuntu 16.04 LTS with ROS Kinetic), those exact steps build the package. You can also use `catkin build` from [catkin tools](https://catkin-tools.readthedocs.io/en/latest/).

To use the package, please consult the [Tutorials](http://wiki.ros.org/apriltags2_ros/Tutorials).

dmalyuta, 2018-02-28:
Did this answer your question? I will close the issue then.

closed 2018-03-14 by dmalyuta