AprilRobotics/apriltag_ros #3: position not shown in /tag_detections

SietseAchterop, 2018-03-13:
When using continuous_detection the tag is detected and the outline is shown in the image, so it is detected.
 But the values of the pose are NOT updated in /tag_detections, they stay all on zero.
What am I doing wrong?

JiekeLi, 2018-03-14:
I  met the same provlem.I just follow the tutorials.

JiekeLi, 2018-03-14:
I think I have found the reason: when I echo the topic--/camera/camera_info ,I found that all ths parameters in camera_info are zero!

dmalyuta, 2018-03-14:
The [getRelativeTransform](https://github.com/dmalyuta/apriltags2_ros/blob/master/apriltags2_ros/src/common_functions.cpp#L440) function computes the tag pose. If your `camera_info` topic has all zero parameters like @JiekeLi  says, then that will indeed play badly with the `cv::solvePnP` function. @SietseAchterop are your parameters all zero for `camera_info`? If so, fix that first (by filling out the K matrix in that topic with your x, y focal lengths and optical center coordinates) and let us know if this fixes your issue.

SietseAchterop, 2018-03-16:
I use the usb_cam package, and that has all the zero's. 
After setting sensible calibration values the camera_info is filled in, and continuous_detection is working as advertised!
Maybe a warning about the camera_info would be helpful.
Thanks for the help, issue solved!

dmalyuta, 2018-03-16:
I'll make a warning about it, thank you.

closed 2018-03-16 by dmalyuta