AprilRobotics/apriltag_ros #10: issue on Raspberry Pi

mickeyouyou, 2018-05-05:
`catkin_make_isolated` on Raspberry Pi result:
```
[ 66%] Linking CXX shared library /home/hdwl/catkin_ws/devel_isolated/apriltags2_ros/lib/libcommon.so
[ 66%] Built target common
Scanning dependencies of target apriltags2_ros_single_image_client_node
Scanning dependencies of target continuous_detector
Scanning dependencies of target single_image_detector
[ 70%] Building CXX object CMakeFiles/continuous_detector.dir/src/continuous_detector.cpp.o
[ 73%] Building CXX object CMakeFiles/apriltags2_ros_single_image_client_node.dir/src/apriltags2_ros_single_image_client_node.cpp.o
[ 76%] Building CXX object CMakeFiles/single_image_detector.dir/src/single_image_detector.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
CMakeFiles/single_image_detector.dir/build.make:62: recipe for target 'CMakeFiles/single_image_detector.dir/src/single_image_detector.cpp.o' failed
make[2]: *** [CMakeFiles/single_image_detector.dir/src/single_image_detector.cpp.o] Error 4
CMakeFiles/Makefile2:1387: recipe for target 'CMakeFiles/single_image_detector.dir/all' failed
make[1]: *** [CMakeFiles/single_image_detector.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

could you give some advise to build it ?

YuehChuan, 2018-06-26:
@mickeyouyou  It is running out the memory. Usually ghappen in RAM constrained embedded board like raspbeery pi. 
Solution: plug in an USB stick and enlarge SWAP memory 
by 
```bash=
$ sudo mkswap /dev/sda1 
$ sudo swapon /dev/sda1
```
sda1 is where your usb mounting point. 

reference how I build ROS2 on raspberry pi3
https://hackmd.io/s/BysaB355-#

mickeyouyou, 2018-07-03:
thank you so much for your reply

closed 2019-05-13 by wxmerkt