AprilRobotics/apriltag_ros #13: Covariance set to zero.

lcluz, 2018-05-11:


wxmerkt, 2019-05-13:
Closing due to inactivity - can you please reopen with more details if the question still persists?

closed 2019-05-13 by wxmerkt

graulef, 2020-04-15:
I believe the issue addressed here is that the covariance matrix provided in the PoseWithCovarianceStamped message is never filled with any non-zero values. I just had a look at the source code and could not find anything that would compute the pose covariance. Therefore, one of two things should be done

1. Compute the covariance and provide in the message (not sure how this would / coud be done).
2. Use the message type PoseStamped instead to avoid confusion.

I hope this clarifies the issue at hand. Cheers!