AprilRobotics/apriltag_ros #16: No default tags description

davelkan, 2018-05-14:
The current set up assumes all tags that may be detected are accounted for directly in a  config file. This does not support applications involving large number of standardized tags, which is very common in industry. There should be a default description or size that can be optionally activated as a catch-all setting for detected tags.

kevin-cognite, 2021-05-05:
Hi,

Are there any new updates/progress on this issue? 

I'm very interested in using this in industrial applications, where keeping track of all the tags deployed in the field in a config file can be tedious/unfeasible.

I would also appreciate any suggestions/workaround to this issue. Thanks!

rgreid, 2021-05-05:
Sounds like a useful enhancement.  Are you or your team able to implement this and do a PR?  Can I suggest you bounce design ideas around here so the PR is accepted? Cheers.