AprilRobotics/apriltag_ros #17: TODO treat issues and pull requests

dmalyuta, 2018-05-14:
I will go over all existing issues and pull requests in June/July (currently my time is spent on research and university studies). Thank you from the bottom of my heart to everyone for participating!

wxmerkt, 2018-10-15:
Hi @dmalyuta,
Thank you very much for your hard work in making and releasing this AprilTags2 ROS wrapper! We are heavily using it and would love to see many of the PRs being merged in annd the wrapper released a a binary - and are happy in providing maintenance support to do so if desired or requested.

Again, thanks for your great work and please let us know if we can help :)

dmalyuta, 2018-10-15:
@wxmerkt Thanks for the kind words! Definitely, if you'd like to become a collaborator and can help with the PRs, let me know and I'll add you.

wxmerkt, 2018-10-16:
Awesome, thank you very much. We are testing a version with #4, #29, and #34 merged in and it works well for long operation times (hours) on Kinetic.

wxmerkt, 2018-10-16:
Thank you very much for the invitation to collaborate, can I start to review and accept pull requests that work?

dmalyuta, 2018-10-16:
Yes, please feel free to do that.

wxmerkt, 2018-10-16:
Awesome, thank you :-)

closed 2019-05-13 by wxmerkt