AprilRobotics/apriltag_ros #20: Extremely High CPU Usage

davelkan, 2018-05-15:
A key advantage of Apriltags 2 over the original is supposedly the substantial savings in CPU. However i'm observing approximately 8x more CPU usage by apriltags2 then by a previous implementation of the original apriltags algorithms(ported to C++) here: https://github.com/swatbotics/apriltags-cpp

dmalyuta, 2018-05-17:
David, thank you for all the activity! I'll get back to you on all points next month. Just wanted to say so that you don't feel like you're writing into the void.

davelkan, 2018-05-18:
yep i saw your announcement. There may be a few more PRs from me between now and then :+1: 

thanks for putting this library together. I'm investigating the CPU usage in comparison to the swatbotics apriltags implementation and i'll let you know what i find.

davelkan, 2018-06-04:
I think the key issue with the CPU is the thresholding. Far too many canidate clusters are created. With better thresholding we may be able to reduce CPU usage dramatically

davelkan, 2018-07-06:
@dmalyuta 
I've been working on incorporating opencv in order to speed up the code, which includes a partial C++ conversion. Areyou interested in having this merged back?

mhkabir, 2018-07-10:
@davelkan Would love to test your changes if you pushed them to a branch somewhere! I'm working on switching my old software stack to use AprilTags2, and would be happy to help however I can. :)

davelkan, 2018-07-12:
I'll work on cleaning them up this week. I've made fairly dramatic changes to get this compatible with my set up. I'll try and get this up by end of week.

dmalyuta, 2018-08-06:
@davelkan I'm happy for you to push all your work into a separate branch and we can take it from there to eventually merge things with master!

skohlbr, 2018-08-07:
There's also [discussion on ROS Answers about high observed CPU usage](https://answers.ros.org/question/299115/april-tag-2-ros-pose-detection/). I linked to this issue in the comments.

davelkan, 2018-08-15:
@dmalyuta sorry for the delay. I've had a bunch of other priority work dumped on my lap and have been sidetracked. I'm in the process of getting my various changes through an internal review and will be pushing it to a branch here as soon as possible

rjosodtssp, 2019-03-07:
Anyone solve this problem yet ? It really uses too much CPU on my computer

wxmerkt, 2019-03-07:
Can you please try pull request #43? I think this should fix your issue.

PR #43 (2019-05-13): Upgrade to AprilTag 3, fix installation, and performance improvements
This is quite a big pull request and incorporates many other currently open pull requests. I hope to address both installation and performance issues with this one. 

- Upgrades to AprilTags 3, and in specific a thin CMake overlay of the now-officially-on-GitHub version. This replaces the apriltags2 catkin wrapper which was a fork and not fully working (installation...). AprilTags 3 should be faster by itself.
- As part of this, some tag families and options are gone (refine_pose, black_border, refine_decode)
- Renamed the l [... cut]
closed 2019-05-13 by wxmerkt