AprilRobotics/apriltag_ros #23: Stops working after a while

lcluz, 2018-06-06:
Hello. 
I am running apriltags2_ros all the time on my mobile and after a while it just stops working, no crash or anything. Just image doesn't show and detections are not there.
The only different thing is that when i kill the launch it says "[apriltags2_ros_continuous_node-14] escalating to SIGTERM ", which doesn't appear when it is running normally.

Thanks, 
Luís

SHERLOCKLS, 2018-06-11:
we also find the same problem.
And the log shows that it received the signal  SigsegV and caused segmentation fault. But we didn't figure out the reason and still work on it.

davelkan, 2018-06-11:
@lcluz @SHERLOCKLS What resolution images are you using? There may be an issue with zarray set up that causes failures when extremely large clusters are detected (~3000 points). I'm having some failures that may be related.

SHERLOCKLS, 2018-06-14:
We use an image with a resolution of 1280x720. Is that too large?

davelkan, 2018-06-19:
No I think this is a bug, potentially with memory allocation. I'm not sure. But the issue i'm seeing would be more common with higher resolutions. Try lowering the resolution and see if the if the issue is less frequent. If it is this may be the same issue i'm seeing.

wxmerkt, 2019-03-07:
Can you please try pull request #43? This _may_ fix your issue.

wxmerkt, 2019-05-13:
Closing due to inactivity and that the bug may be fixed with the new release - if it persists, please reopen.

closed 2019-05-13 by wxmerkt