AprilRobotics/apriltag_ros #28: how can I see the position of apriltags?

niteofhunter, 2018-07-31:
when I use it, the /tag_detections_image is work well.
but when I use /tag_detections, it detect id and size but it not detect position

header: 
  seq: 7346
  stamp: 
    secs: 1533046829
    nsecs: 846270190
  frame_id: "usb_cam"
detections: 
  - 
    id: [3]
    size: [0.1]
    pose: 
      header: 
        seq: 7348
        stamp: 
          secs: 1533046829
          nsecs: 846270190
        frame_id: "usb_cam"
      pose: 
        pose: 
          position: 
            x: 0.0
            y: 0.0
            z: 0.0
          orientation: 
            x: 0.0
            y: 0.0
            z: 0.0
            w: 1.0
        covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  - 
    id: [0, 4, 3, 2, 1]
    size: [0.1, 0.1, 0.1, 0.1, 0.1]
    pose: 
      header: 
        seq: 7348
        stamp: 
          secs: 1533046829
          nsecs: 846270190
        frame_id: "usb_cam"
      pose: 
        pose: 
          position: 
            x: 0.0
            y: 0.0
            z: 0.0
          orientation: 
            x: 0.0
            y: 0.0
            z: 0.0
            w: 1.0
        covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

how can I know the position of apriltag?

MedericFourmy, 2019-01-14:
I have the same problem here, trying to solve it.

e-guy, 2019-02-28:
I met the same issue, how can i solve it?

bankya, 2019-03-09:
Hey @e-guy , any chance you were able to solve this? I'm having the same problem

e-guy, 2019-03-10:
As what Medric said，you need set your camera intrinsic matrix，I set it in the source code

wxmerkt, 2019-05-13:
Closing as this issue appears resolved - if not, please reopen.

closed 2019-05-13 by wxmerkt