AprilRobotics/apriltag_ros #36: Launch files not installed

CameronDevine, 2018-11-06:
When attempting to build `apriltags2_ros` the launch files are not installed into the catkin workspace path. To install I ran,

```bash
cd ~/catkin_ws/src
git clone https://github.com/dmalyuta/apriltags2_ros.git
cd ..
catkin_make
catkin_make install
source install/setup.bash
```

When I attempt to run `roslaunch apriltag2_ros continuous_detection.launch` I get an error message that `continuous_detection.launch` is not a part of `apriltag2_ros`. Using the `find` command I can tell that no launch files from `apriltag2_ros` are in my `~/catkin_ws/install` directory.

ceplm, 2019-02-21:
The package is apriltags2_ros, you misspelled it.

wxmerkt, 2019-02-21:
I think this is a valid issue - quick glance at the CMakeLists.txt shows no install directives. Do you mind proposing a pull request? Thanks!

ceplm, 2019-02-21:
Sorry, I just only installed the package using code above and tried launching it. Since no one commented, I thought the misspelled name could be issue.

CameronDevine, 2019-02-21:
@wxmerkt I should be able to create a pull request sometime within the next week or so.

wxmerkt, 2019-02-21:
Thank you @CameronDevine

CameronDevine, 2019-02-25:
@wxmerkt, I am unsure of which files should be installed. I know the files in the [launch](https://github.com/dmalyuta/apriltags2_ros/tree/master/apriltags2_ros/launch) and [config](https://github.com/dmalyuta/apriltags2_ros/tree/master/apriltags2_ros/config) directories should be installed, but I am unsure about the files in the [scripts](https://github.com/dmalyuta/apriltags2_ros/tree/master/apriltags2_ros/scripts) directory. I believe [analyze_image.sh](https://github.com/dmalyuta/apriltags2_ros/blob/master/apriltags2_ros/scripts/analyze_image.sh) could be run with `rosrun`, but [calibrate_bundle.m](https://github.com/dmalyuta/apriltags2_ros/blob/master/apriltags2_ros/scripts/calibrate_bundle.m) could not because it is a Matlab script. Should either of these two files be installed?

wxmerkt, 2019-02-25:
I don't think so. The `analyze_image.sh` is a convenience wrapper around another roslaunch call so I'd rather not include it.

CameronDevine, 2019-03-04:
I submitted #42 tonight which fixes this issue for me.

wxmerkt, 2019-03-07:
Thank you for the pull request. I have incorporated along many others into a full upgrade of the library. Please try #43 whether this fixes your issue.

wxmerkt, 2019-05-13:
This should have been fixed by #43.

closed 2019-05-13 by wxmerkt