AprilRobotics/apriltag_ros #38: ROS Warning - Tag descriptions not found.

ColmP, 2018-12-19:
Hello,
I just installed apriltags2 for ROS Melodic. Apart from updating the camera topic information in the continuous_detection.launch file, none of the other settings have been modified.

When I have a Tag36h11 tag in the field of view  I get the following warning:
[ WARN] [1545183966.758860475]: Requested description of standalone tag ID [0], but no description was found...

/tag_detections_image is being published (without the tag highlighted). /tag_detections returns an empty detections array.

Camera data is being published by the usb_cam node to topic /usb_cam/image_raw ( and /usb_cam/camera_info).

I am guessing that this is a rookie issue and any help would be greatly appreciated.

Thank you,
Colm.

wxmerkt, 2018-12-19:
@ColmP: Have you defined the tags you are looking for in the `config/tags.yaml`?

https://github.com/dmalyuta/apriltags2_ros/blob/44ecc217477c7ad3dbcce0a1a375871809a7c60f/apriltags2_ros/config/tags.yaml#L15-L19

^ Define the ID and size of tags here along with a name and it will publish accordingly :-)

ColmP, 2018-12-19:
Hi Wolfgang,
That was the problem, I figured that it was a rookie mistake on my end. I have it working now.
Thank you for your help.
Best Regards,
Colm

closed 2018-12-19 by ColmP