AprilRobotics/apriltag_ros #41: AprilTags size

aleixpb2, 2019-02-21:
This is a comment more than an issue. I think the apriltags size is the side of the tag in meters including the padding bits on each side since when I follow the instructions on [Printing Your Own AprilTags](http://wiki.tekkotsu.org/index.php/AprilTags#Printing_Your_Own_AprilTags) the size I specify programmatically includes those bits when I measure it in real life. Could you add it to your documentation page http://wiki.ros.org/apriltags2_ros/Tutorials/Detection%20in%20a%20video%20stream when explaining the file config/tags.yaml? If I am wrong, please let me know.

Thank you,
Aleix

FJarecki, 2019-05-09:
"padding bits" you mean black borded (1 bit) or addidtional white border around black border? :)
I am asking cause i dont know which size we give in tags.yaml? Apriltag with white and black border or only with black border?

aleixpb2, 2019-05-09:
I was referring to the additional white border. I think it's included in the size of the tag. Therefore, you can consider that the tags' size is actually 10 bits.

FJarecki, 2019-05-10:
In this article: https://doc.rc-visard.com/latest/en/tagdetect.html i found note: "During pose estimation, the tag’s size is also estimated, while assuming the tag to be square. For QR codes, the size covers the full tag. For AprilTags, the size covers only the black part of the tag, hence ignoring the outermost white border." and I assumed that in this algorithm is the same. I will try with different sizes of this white border and it will return different poses of tag (the algorithm should think that the marker is further / closer)

Jakubach, 2019-06-29:
Hello,
I was also looking for this information so I found an answer from april-tag support:
[[april-support] AprilTag Size](https://april.eecs.umich.edu/pipermail/april-support/2017-May/000032.html)
They say, that "tag size" mean everything without outer, white border. 
Apriltag_ros currently is a wrapper for a AprilTag3 [linked here](https://github.com/AprilRobotics/apriltag)

aleixpb2, 2019-09-05:
Yes, it makes sense that it's only counting the black border. Thanks!

closed 2019-09-05 by aleixpb2