AprilRobotics/apriltag_ros #45: Rename repo to apriltag_ros to align with upstream Apriltag library

rgreid, 2019-04-05:
Continuing the discussion from [here](https://github.com/dmalyuta/apriltags2_ros/pull/43#issuecomment-479955663)

The upstream repo https://github.com/AprilRobotics/apriltags uses `apriltag` consistently in CMakeLists.txt and package.xml and `libapriltag` in the Makefile.  This repo uses `apriltag` also. 

For consistency, this repo/package should _probably_ be renamed `apriltag_ros`? 

Before renaming, we should probably figure how it fits with the broader ROS community.
 - http://wiki.ros.org/apriltags_ros  hasn't been updated since Indigo
 - http://wiki.ros.org/apriltags2_ros points to this repo

@dmalyuta how did you create the ROS wiki page?  If this package is renamed to http://wiki.ros.org/apriltag_ros, can redirects be added for the other pages?

wxmerkt, 2019-04-05:
Thank you for starting this conversation! I think we can maintain a tag/branch with the current master as `apriltags2_ros` and continue development of the AprilTag3 + future versions within this repository/master.

I also think renaming to `apriltag_ros` within this repo (or by transferring it to another organisation?) would be the most consistent. While renaming to `apriltag_ros` could lead to confusion with the indigo-era `apriltags_ros` (which is no longer maintained), if we only target kinetic/melodic/newer it should be okay?

dmalyuta, 2019-04-05:
@rgreid I am fine with either renaming or forking the repo, whichever would be best for the community. Regarding the wiki, [it looks like it can be renamed](https://answers.ros.org/question/33571/moving-wiki-pages/).

rgreid, 2019-04-07:
If we rename this repository to `apriltag_ros`, then Github will create a redirect from `apriltags2_ros`.  Anyone with builds pointing to the old repo URL will get the `master` version, i.e. they'll soon be upgraded to Apriltags 3, which may break things.   All said, I'm fine with that!

@wxmerkt which organisation would you suggest transferring the repo to? We could ask @mkrogius about https://github.com/AprilRobotics/apriltag_ros 

I did a quick search and couldn't see if Github adds redirects for personal repos transferred to organisations. Anyone know? 

@mkrogius what's your thoughts?

wxmerkt, 2019-04-07:
> If we rename this repository to apriltag_ros, then Github will create a redirect from apriltags2_ros. Anyone with builds pointing to the old repo URL will get the master version, i.e. they'll soon be upgraded to Apriltags 3, which may break things. All said, I'm fine with that!

Sadly, anything short of a hard fork does reduce the feature set that GitHub usually provides (search in repository, comparisons etc.). We can check statistics/insights how many unique clones the repository got recently to evaluate how much of an impact this could be.

Organisation-wise, maybe @aprilrobotics or @ros-perception would be great. Or a new one @apriltag_ros/apriltag_ros.

I previously transferred personal -> org -> personal and redirects were maintained.

mkrogius, 2019-04-08:
I would be happy to host this repo in AprilRobotics and give access to the current maintainers. But, I also think that it would be reasonable to keep this repo in the current org since @dmalyuta has put the work in to build the thing. I'm good with whatever people decide since I don't know anything about current practices for ROS releases.

dmalyuta, 2019-04-08:
I would be fine with whatever decision that would work better for the community. If hosting on @AprilRobotics would give wider and more straightforward access to people, I am happy with that. It would be nice though if the project keeps a request to cite my thesis/journal paper (currently in review), if people use the code in their research.

mkrogius, 2019-04-11:
@rgreid Yes, this is supported by Ed and the rest of the April Lab

rgreid, 2019-04-11:
@mkrogius Great, good to know! Say "Hi" to Ed from me!

Referring to this issue https://github.com/AprilRobotics/apriltags/issues/16, how about two repos/packages named consistently?
 - https://github.com/AprilRobotics/apriltag
 - https://github.com/AprilRobotics/apriltag_ros

 @dmalyuta I feel like this would be a good move for both you and the community (maximizing adoption of your code, while removing confusion about support for the various Apriltag ROS packages). 

Again, just my two cents!

wxmerkt, 2019-04-11:
For the latter point - if everybody and especially @dmalyuta agrees, it would be great to rename/transfer this repository to @AprilRobotics to ensure all issues/PR/etc are transferred over and continuity. I am happy to continue to volunteer as maintainer to review/merge PRs and address issues.

dmalyuta, 2019-04-11:
@rgreid @wxmerkt As I said, I'm more than happy (and honored) to have the repo transferred to @AprilRobotics. I think that it is a great decision.

mkrogius, 2019-04-11:
I've renamed the AprilRobotics/apriltags repo to AprilRobotics/apriltag. I also found these instructions for transferring repos: https://help.github.com/en/articles/transferring-a-repository#transferring-a-repository-owned-by-your-personal-account

dmalyuta, 2019-04-11:
@mkrogius Do I have the go-ahead from everyone to transfer the repo to @AprilRobotics as per those instructions? I'll give it a day and if no one gives a "no go" in this thread then I'll do the transfer.

dmalyuta, 2019-04-13:
@mkrogius I tried to make the transfer to @AprilRobotics as per the instructions you linked, but got ``You don’t have the permission to create repositories on AprilRobotics``. I suppose that you need to add me to the organization?

mkrogius, 2019-04-15:
@dmalyuta I sent you a member invite for AprilRobotics, hope that works

dmalyuta, 2019-04-15:
Just made the transfer. Let me know if there are any issues.

rgreid, 2019-04-15:
Looks like that worked! Do you have permissions to rename the repo too?

mkrogius, 2019-04-17:
I went ahead and renamed the repo. It looks like there are still references to apriltags2_ros inside the repo which should be cleaned up

wxmerkt, 2019-04-17:
@mkrogius Thank you. I will proceed with a clean-up and update #43 correspondingly.

rgreid, 2019-04-20:
Two thoughts before closing this ticket out:
1. Add version tags as per my comment here https://github.com/AprilRobotics/apriltag_ros/pull/43
1. Need to update the ROS wiki. @wxmerkt did I see in one of your commits http://wiki.ros.org/apriltag_ros?

wxmerkt, 2019-04-21:
Re (1): I've already tagged the current master as 1.0.0 and will do so after merge of #43 as 2.0.0 (cf. https://github.com/AprilRobotics/apriltag_ros/releases). Happy to follow your suggestion from #43 to do 2.0/3.0.

Re (2): I don't have permissions for that and haven't done that previously - happy to take any pointers and do it.

dmalyuta, 2019-04-22:
I did just rename the Wiki page: see [http://wiki.ros.org/apriltag_ros](http://wiki.ros.org/apriltag_ros). However, it needs the text updated for consistency. For the wiki page header, I believe that the steps on this page need to be followed: [http://wiki.ros.org/rosdistro/Tutorials/Indexing%20Your%20ROS%20Repository%20for%20Documentation%20Generation](http://wiki.ros.org/rosdistro/Tutorials/Indexing%20Your%20ROS%20Repository%20for%20Documentation%20Generation)

rgreid, 2019-05-02:
Quick update: I've renamed the ROS Wiki tutorial pages and inserted banners at the top of all the old/unsupported pages e.g. http://wiki.ros.org/apriltags_ros etc.  The main http://wiki.ros.org/apriltag_ros page still needs updating. 

@wxmerkt: Initial tests on https://github.com/AprilRobotics/apriltag_ros/pull/43 are looking good. I'll run some more tomorrow.

PR #43 (2019-05-13): Upgrade to AprilTag 3, fix installation, and performance improvements
This is quite a big pull request and incorporates many other currently open pull requests. I hope to address both installation and performance issues with this one. 

- Upgrades to AprilTags 3, and in specific a thin CMake overlay of the now-officially-on-GitHub version. This replaces the apriltags2 catkin wrapper which was a fork and not fully working (installation...). AprilTags 3 should be faster by itself.
- As part of this, some tag families and options are gone (refine_pose, black_border, refine_decode)
- Renamed the l [... cut]
closed 2019-05-13 by wxmerkt