AprilRobotics/apriltag_ros #46: Any limit on pixel_format and size?

PatronusCoder, 2019-04-30:
Hey

I have been trying to use AprilTag detection with a GigE camera that gives me an output of 1280x720 image size and a BayerRG8 image format. It works but the detection rate drops considerably. The rectified image is being published at ~60Hz and the tf information at ~10Hz. When I reduced the image size to 640x480, the update rate became better. I even tested this with a USB camera with 30FPS, 1280x720 and mjpeg format, and found similar results. So is the implementation limited in performance by these factors? Also, what has been the highest publishing frequency for this implementation ever noted and for what parameters? For me it has been max. 30Hz for 640x480 size with yuyv format.

wxmerkt, 2019-05-13:
Hi,
I think performance is highly dependent on your system setup and whether you use message publishing or a nodelet. It is expected that the detection rate to drop with larger image sizes (as processing takes longer), but I do not have numbers on that I am afraid.

degpr7rng, 2019-05-13:
Hey @wxmerkt . I checked out other open and closed issues in this repository and found that some other guys also submitted issues concerning high CPU usage and low detection rate. To this, you suggested the pull request #43 and I pulled this repository after you merged it with the master branch. But still the detection rate is no more than 4Hz. I am using I7 octa core processor with image size 1936x1216. I am getting raw images at ~24Hz and rectified image at about the same frequency. But one thing I noticed was there were multiple threads running from image_proc node. Could this be a reason behind the slow performance?

wxmerkt, 2019-05-13:
Hi @degpr7rng,
This is quite a large image size! What percentage of the CPU is used by the image_rect and other nodes? Are you using the nodelet or the individual node? Do you set any special cmake args or parameters for the workspace?

degpr7rng, 2019-05-14:
[image]

Hey @wxmerkt. I have added a screenshot showing the CPU percentage each node is taking. I am using nodes and have set the following CMake args for my compiler:

-DCMAKE_CXX_FLAGS=-Wall -march=native -O2 -ggdb

MrBoriska, 2019-07-12:
To be continued...

wxmerkt, 2020-10-21:
Closing due to old age - please reopen if issue still persists

closed 2020-10-21 by wxmerkt