AprilRobotics/apriltag_ros #48: Linking problem

davizinho5, 2019-06-01:
Hello, 

I'm trying to build this project in Ubuntu 18.04 and ROS-melodic. 

I just did: 
(i) cd catkin_ws/src/
(ii) git clone https://github.com/AprilRobotics/apriltag_ros.git
(iii) cd catkin_ws/ & catkin_make

I'm getting a linking problem against a very basic ROS library:

```
[ 96%] Linking CXX executable /home/das/catkin_ws/devel/lib/apriltag_ros/apriltag_ros_single_image_client_node
CMakeFiles/apriltag_ros_single_image_client_node.dir/src/apriltag_ros_single_image_client_node.cpp.o: In function `main':
apriltag_ros_single_image_client_node.cpp:(.text.startup+0x101): undefined reference to `ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__debug::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
apriltag_ros_single_image_client_node.cpp:(.text.startup+0x1b0): undefined reference to `ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__debug::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
collect2: error: ld returned 1 exit status

```

Any idea why?

rgreid, 2019-06-02:
Can you use `catkin build` instead? Read why `catkin-tools` is preferred here:  https://catkin-tools.readthedocs.io/en/latest/migration.html  Otherwise, what does `catkin config` say?

davizinho5, 2019-06-02:
Understood, I'll use catkin build from now on. 
I cannot make the "Step-by-step migration provided in the web, since my packages do not build with catkin_make. However, I did `catkin build` as you asked. 
The same problem applies to the all the nodes in the package. 

```
Profile:                     default
Extending:          [cached] /home/das/catkin_ws/devel:/opt/ros/melodic:/home/das/sources/cmake-modules:/home/das/sources/fcl/install:/usr/lib/cmake/eigen3/:/home/das/sources/opencv-4.1.0/install/
Workspace:                   /home/das/catkin_ws
------------------------------------------------------------------------------------------
Build Space:        [exists] /home/das/catkin_ws/build_cb
Devel Space:        [exists] /home/das/catkin_ws/devel_cb
Install Space:      [unused] /home/das/catkin_ws/install_cb
Log Space:          [exists] /home/das/catkin_ws/logs_cb
Source Space:       [exists] /home/das/catkin_ws/src
DESTDIR:            [unused] None
------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------------------------------------
[build] Found '8' packages in 0.0 seconds.                                               
[build] Package table is up to date.                                                     
Starting  >>> apriltag_ros                                                               
Starting  >>> apriltags                                                                  
Starting  >>> reflex_msgs                                                                
Starting  >>> reflex_visualizer                                                          
Finished  <<< apriltags                          [ 0.2 seconds ]                         
Finished  <<< reflex_msgs                        [ 0.6 seconds ]                         
Finished  <<< reflex_visualizer                  [ 0.6 seconds ]                         
Starting  >>> control_node                                                               
Starting  >>> reflex                                                                     
Starting  >>> reflex_driver                                                              
Finished  <<< reflex_driver                      [ 0.2 seconds ]                         
Finished  <<< control_node                       [ 0.4 seconds ]                         
Finished  <<< reflex                             [ 0.6 seconds ]                         
Starting  >>> reflex_manipulation                                                        
_________________________________________________________________________________________
Errors     << apriltag_ros:make /home/das/catkin_ws/logs_cb/apriltag_ros/build.make.002.log
CMakeFiles/apriltag_ros_single_image_client_node.dir/src/apriltag_ros_single_image_client_node.cpp.o: In function `main':
apriltag_ros_single_image_client_node.cpp:(.text.startup+0x101): undefined reference to `ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__debug::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
apriltag_ros_single_image_client_node.cpp:(.text.startup+0x1b0): undefined reference to `ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__debug::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/das/catkin_ws/devel_cb/.private/apriltag_ros/lib/apriltag_ros/apriltag_ros_single_image_client_node] Error 1
make[1]: *** [CMakeFiles/apriltag_ros_single_image_client_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/apriltag_ros_continuous_node.dir/src/apriltag_ros_continuous_node.cpp.o: In function `main':
apriltag_ros_continuous_node.cpp:(.text.startup+0x22d): undefined reference to `nodelet::Loader::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_strin [... cut ...]
Finished  <<< reflex_manipulation                [ 0.1 seconds ]                         
[build] Summary: 7 of 8 packages succeeded.                                              
[build]   Ignored:   None.                                                               
[build]   Warnings:  None.                                                               
[build]   Abandoned: None.                                                               
[build]   Failed:    1 packages failed.                                                  
[build] Runtime: 1.7 seconds total.
```

rgreid, 2019-06-02:
I'm not a catkin expert, however the `Extending` line above looks odd. I'd try a clean workspace and explicitly chain only the ROS workspace:
```
catkin clean
catkin config --extend /opt/ros/melodic
```
Read about extending workspaces [here](https://catkin-tools.readthedocs.io/en/latest/verbs/catkin_config.html#explicitly-specifying-workspace-chaining), note the difference in their `Extending` line.

davizinho5, 2019-06-03:
I did that. The `Extending` line now only has the /opt/ros/melodic. 
However, I still get the same error. 

```
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/das/catkin_ws
-----------------------------------------------------------
Build Space:        [exists] /home/das/catkin_ws/build_cb
Devel Space:        [exists] /home/das/catkin_ws/devel_cb
Install Space:      [unused] /home/das/catkin_ws/install_cb
Log Space:          [exists] /home/das/catkin_ws/logs_cb
Source Space:       [exists] /home/das/catkin_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '8' packages in 0.0 seconds.                                                                        
[build] Package table is up to date.                                                                              
Starting  >>> apriltag_ros                                                                                        
Starting  >>> apriltags                                                                                           
Starting  >>> reflex_msgs                                                                                         
Starting  >>> reflex_visualizer                                                                                   
Finished  <<< apriltags                          [ 0.2 seconds ]                                                  
Finished  <<< reflex_visualizer                  [ 0.6 seconds ]                                                  
Finished  <<< reflex_msgs                        [ 0.6 seconds ]                                                  
Starting  >>> control_node                                                                                        
Starting  >>> reflex                                                                                              
Starting  >>> reflex_driver                                                                                       
Finished  <<< reflex_driver                      [ 0.3 seconds ]                                                  
Finished  <<< control_node                       [ 0.5 seconds ]                                                  
Finished  <<< reflex                             [ 0.6 seconds ]                                                  
Starting  >>> reflex_manipulation                                                                                 
__________________________________________________________________________________________________________________
Errors     << apriltag_ros:make /home/das/catkin_ws/logs_cb/apriltag_ros/build.make.002.log                       
CMakeFiles/apriltag_ros_single_image_client_node.dir/src/apriltag_ros_single_image_client_node.cpp.o: In function `main':
apriltag_ros_single_image_client_node.cpp:(.text.startup+0x101): undefined reference to `ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__debug::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
apriltag_ros_single_image_client_node.cpp:(.text.startup+0x1b0): undefined reference to `ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__debug::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/das/catkin_ws/devel_cb/.private/apriltag_ros/lib/apriltag_ros/apriltag_ros_single_image_client_node] Error 1
make[1]: *** [CMakeFiles/apriltag_ros_single_image_client_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/apriltag_ros_continuous_node.dir/src/apriltag_ros_continuous_node.cpp.o: In function `main':
apriltag_ros_continuous_node.cpp:(.text.startup+0x22d): undefined reference to `nodelet::Loader::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<cha [... cut ...] or<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/das/catkin_ws/devel_cb/.private/apriltag_ros/lib/apriltag_ros/apriltag_ros_single_image_server_node] Error 1
make[1]: *** [CMakeFiles/apriltag_ros_single_image_server_node.dir/all] Error 2
make: *** [all] Error 2
```

wxmerkt, 2019-06-03:
I am wondering whether we forgot a dependency - can you try building and linking with a single thread? 

```catkin build -j1 -l1```

davizinho5, 2019-06-03:
Looks like the same to me:

```
catkin build -j1 -l1
-----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/das/catkin_ws
-----------------------------------------------------------
Build Space:        [exists] /home/das/catkin_ws/build_cb
Devel Space:        [exists] /home/das/catkin_ws/devel_cb
Install Space:      [unused] /home/das/catkin_ws/install_cb
Log Space:          [exists] /home/das/catkin_ws/logs_cb
Source Space:       [exists] /home/das/catkin_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        -j1 -l1
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '8' packages in 0.0 seconds.                                                                        
[build] Package table is up to date.                                                                              
Starting  >>> apriltag_ros                                                                                        
__________________________________________________________________________________________________________________
Errors     << apriltag_ros:make /home/das/catkin_ws/logs_cb/apriltag_ros/build.make.004.log                       
CMakeFiles/apriltag_ros_single_image_client_node.dir/src/apriltag_ros_single_image_client_node.cpp.o: In function `main':
apriltag_ros_single_image_client_node.cpp:(.text.startup+0x101): undefined reference to `ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__debug::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
apriltag_ros_single_image_client_node.cpp:(.text.startup+0x1b0): undefined reference to `ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__debug::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/das/catkin_ws/devel_cb/.private/apriltag_ros/lib/apriltag_ros/apriltag_ros_single_image_client_node] Error 1
make[1]: *** [CMakeFiles/apriltag_ros_single_image_client_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/das/catkin_ws/build_cb/apriltag_ros; catkin build --get-env apriltag_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................
Failed     << apriltag_ros:make                  [ Exited with code 2 ]                                           
Failed    <<< apriltag_ros                       [ 1.0 seconds ]                                                  
Abandoned <<< apriltags                          [ Unrelated job failed ]                                         
Abandoned <<< reflex_msgs                        [ Unrelated job failed ]                                         
Abandoned <<< reflex_visualizer                  [ Unrelated job failed ]                                         
Abandoned <<< control_node                       [ Unrelated job failed ]                                         
Abandoned <<< reflex                             [ Unrelated job failed ]                                         
Abandoned <<< reflex_driver                      [ Unrelated job failed ]                                         
Abandoned <<< reflex_manipulation                [ Unrelated job failed ]                                         
[build] Summary: 0 of 8 packages succeeded.                                                                       
[build]   Ignored:   None.                                                                                        
[build]   Warnings:  None.                                                                                        
[build]   Abandoned: 7 packages were abandoned.                                                                   
[build]   Failed:    1 packages failed.                                                                           
[build] Runtime: 1.1 seconds total.
```

wxmerkt, 2019-06-03:
Hmm, sometimes I've had issues with catkin-tools that went away by removing the `.catkin_tools`, `.private`, `build`, `devel`, `logs` directories in the root of a workspace manually (requiring `catkin config --init` - and setting your options afterwards). Given that it's not failing at the configure stage, `nodelet` appears to be installed.

I just saw you have a package called `apriltags` - which one is that? The library is now `apriltag` without an `s`.

davizinho5, 2019-06-03:
Apriltag (without s) is just an old version I was making tests with. 

Your deleting a starting from scratch worked!!!
I'm guessing the problem was in `.catkin_tools` because I had already tried deleting the others.

Thanks a lot!

closed 2019-06-03 by davizinho5