AprilRobotics/apriltag_ros #49: build error in jetson TX2 (jetpack 4.2 )

amiao1017, 2019-06-11:
My platform is Jetson TX with jetpack 4.2, ros melodic.  When I use catkin build to install apriltag_ros, there is an error.

[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _apriltag_ros_generate_messages_check_deps_AnalyzeSingleImage
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target _apriltag_ros_generate_messages_check_deps_AprilTagDetectionArray
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target _apriltag_ros_generate_messages_check_deps_AprilTagDetection
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
[ 10%] Built target apriltag_ros_generate_messages_lisp
[ 20%] Built target apriltag_ros_generate_messages_cpp
[ 36%] Built target apriltag_ros_generate_messages_py
[ 50%] Built target apriltag_ros_generate_messages_eus
[ 60%] Built target apriltag_ros_generate_messages_nodejs
make[2]: *** 没有规则可制作目标“/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0”，由“/home/amiao1017/catkin_ws/devel/.private/apriltag_ros/lib/libapriltag_ros_common.so” 需求。 停止。
CMakeFiles/Makefile2:2659: recipe for target 'CMakeFiles/apriltag_ros_common.dir/all' failed
make[1]: *** [CMakeFiles/apriltag_ros_common.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 60%] Built target apriltag_ros_generate_messages
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

jacksparkman, 2019-07-17:
Also have a build error on the jetson TX2 (jetpack 4.2 L4T R32.1 (K4.9) Ubuntu 18.04 LTS aarch64)
Tried catkin_make_isolated, below is just the last error message

make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/jack/catkin_ws/devel_isolated/apriltag_ros/lib/libapriltag_ros_common.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 63%] Built target apriltag_ros_generate_messages_eus
Scanning dependencies of target apriltag_ros_generate_messages
[ 63%] Built target apriltag_ros_generate_messages
CMakeFiles/Makefile2:2659: recipe for target 'CMakeFiles/apriltag_ros_common.dir/all' failed
make[1]: *** [CMakeFiles/apriltag_ros_common.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'apriltag_ros': 
  Command '['/home/jack/catkin_ws/devel_isolated/apriltag/env.sh', 'make', '-j6', '-l6']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/jack/catkin_ws/build_isolated/apriltag_ros && /home/jack/catkin_ws/devel_isolated/apriltag/env.sh make -j6 -l6

Command failed, exiting.

wxmerkt, 2019-07-17:
@jacksparkman Do you have OpenCV and all dependencies installed? You can also now get apriltag-ros via apt-get: `sudo apt-get install ros-melodic-apriltag-ros`

jacksparkman, 2019-07-17:
Thanks! apt-get worked for apriltag-ros

wxmerkt, 2019-07-17:
Thank you. I will close this issue. If it re-occurs with all dependencies installed, please reopen.

closed 2019-07-17 by wxmerkt

aydal, 2019-12-20:
Hello @wxmerkt. I am facing the same issue while building on jetson nano. I need the source code to add few more functionalities and hence installing binaries wont work for me. Kindly let me know if have a workaround.
```make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/ayush/catkin_ws/devel/.private/apriltag_ros/lib/libapriltag_ros_common.so'```

wxmerkt, 2019-12-20:
You need to install OpenCV, e.g., from the ROS binaries.

aydal, 2019-12-21:
Have installed cv_bridge and image_proc using binaries. Still getting the error

wxmerkt, 2019-12-22:
Can you run `sudo apt-get install ros-melodic-apriltag-ros` to get all dependencies even if you are compiling from source?