AprilRobotics/apriltag_ros #50: catkin_make error

npd26, 2019-06-12:
Hello,

I am trying to install this packageby cloning it in 
catkin_ws/src and using catkin_make
but it is giving me this error,
```
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:578 (message):
  None of the required 'apriltag' found
```
am I missing something to install before this package?

thanks.

rgreid, 2019-06-12:
@npd26 I added a quickstart to the README.md, you'll find it [here](https://github.com/AprilRobotics/apriltag_ros/tree/rgreid-updated-readme) until the PR is merged.

You need to clone the Apriltag repo also.

closed 2019-06-12 by rgreid

D-aggie, 2019-08-07:
Hello I am still experiencing the same error as the original issue, i tried to follow the link but it is broken, could you please create a new link? Or if there are alternative methods to solve the issue that would work for me too, thank you

npd26, 2019-08-07:
@D-aggie you need to clone also apriltag package as explained here, [https://github.com/AprilRobotics/apriltag_ros/blob/master/README.md](url)
and also use **catkin_make_isolated**