AprilRobotics/apriltag_ros #52: rostopic echo /tag_detections error

npd26, 2019-06-14:
Hi,

After detecting tag I am getting pose information in /tf topic. but when I run 
`rostopic echo /tag_detections` , it is not showing pose

`ERROR: Cannot load message class for [apriltag_ros/AprilTagDetectionArray]. Are your messages built?
`
also it gives me warning when I request topic information,
```
[ WARN] [1560516430.864295787] [/apriltag_ros_continuous_node] [ros.apriltag_ros]: remove_duplicates parameter not provided. Defaulting to true
[ WARN] [1560517507.232646495] [/apriltag_ros_continuous_node] [ros.apriltag_ros]: Requested description of standalone tag ID [573], but no description was found...
```

thanks.

closed 2019-06-14 by npd26

willnil, 2020-01-28:
Hello, 

how did you solve this issue? I am having the same problem as well.

jokla, 2020-01-28:
Have you sourced the file `devel/setup.bash` from the terminal?

`$ source ~/catkin_ws/devel/setup.bash`

willnil, 2020-01-28:
Oh thank you. I installed the Apriltag in devel_isolated and forgot to source the bash file.

Thanks for your quick reply