AprilRobotics/apriltag_ros #55: improve accuracy of detection

npd26, 2019-06-25:
Hello,

I am detecting tag36h11 with realsense camera using only /rgb topic. When I detect tag it is not giving accurate position information w.r.t camera(ex. tag z distance to camera giving me output of 66 cm where in actual it is 63 cm).

so, How can I improve accuracy of this detections? 

is there any usecase where someone also used depth information from realsense with aprtiltag?

thanks.

closed 2019-07-01 by npd26

ajvalenc, 2019-07-14:
I am having the same issue with Kinect sensor (about 3 cm offset). Were you able to solve it? @npd26

npd26, 2019-07-15:
No, but I have found one article [](https://gsyc.urjc.es/jmplaza/papers/jornadasautomatica2016-pnp.pdf)
where it is stated that distance from marker to camera increases, error will also increase.

maximilianwulf, 2019-09-17:
The depth error of the Realsense is approximately 2% of the z value, in this case ~1.26cm. 
The external calibration of the Realsense can degrade, you could try to recalibrate the Realsense camera with [this tool](https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API). The result should be more accurate.