AprilRobotics/apriltag_ros #57: apriltag Pose w.r.t center of tag?

npd26, 2019-07-04:
Hello,

in /tag_detections topic Pose is referred to **center** of apriltag right? or is it top left corner of apriltag?

thanks.

npd26, 2019-07-09:
in this document it is written like [](https://doc.rc-visard.com/latest/en/tagdetect.html),

> The z-axis is pointing “into” the tag. Please note that for AprilTags, although having the white border included in their definition, the coordinate system’s origin is placed exactly at the transition from the white to the black border. Since AprilTags do not have an obvious orientation, the origin is defined as the upper left corner in the orientation they are pre-generated in.

so it should be upper left corner. but in /tag_detections_image we can see center of tag in image_view.

can anyone help?

rgreid, 2019-07-10:
It's w.r.t. the centre. 

TL;DR: This comment can be found in [the source](https://github.com/AprilRobotics/apriltag_ros/blob/master/apriltag_ros/src/common_functions.cpp#L276):

>  these frames together with cv::solvePnP directly avoids
>    // AprilTag 's frames altogether

And the object points that are solved for are distributed `tag_size/2` around 0,0,0 [in the x,y plane here](https://github.com/AprilRobotics/apriltag_ros/blob/master/apriltag_ros/src/common_functions.cpp#L407):
```
objectPoints.push_back(T_oi.get_minor<3, 4>(0, 0)*cv::Vec4d(-s,-s, 0, 1));
objectPoints.push_back(T_oi.get_minor<3, 4>(0, 0)*cv::Vec4d( s,-s, 0, 1));
objectPoints.push_back(T_oi.get_minor<3, 4>(0, 0)*cv::Vec4d( s, s, 0, 1));
objectPoints.push_back(T_oi.get_minor<3, 4>(0, 0)*cv::Vec4d(-s, s, 0, 1));
```

npd26, 2019-07-10:
@rgreid thank you

closed 2019-07-10 by npd26