AprilRobotics/apriltag_ros #66: No rule to make target error

edwardwterry, 2019-11-01:
I tried `catkin build` on apriltag_ros and got this error. 

`make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/username/catkin_ws/devel/.private/apriltag_ros/lib/libapriltag_ros_common.so'.  Stop.
`

I have libopencv_objdetect.so, libopencv_objdetect.so and libopencv_objdetect.so under /usr/lib, but no such objdetect files in /usr/lib/aarch64-linux. When I fake it and copy and rename this file into the specified directory, it just asks for another one, e.g. libopencv_photo.so.3.2.0. I have installed OpenCV 3.4.3 on my Jetson Xavier.

hanzhi713, 2020-01-24:
I resolved this (on ubuntu bionic) by running
```bash
sudo apt install libopencv3.2
```

parakhm95, 2020-03-13:
I think catkin build doesn't work because the core AprilTag library is not a ROS component and is pure C/C++. You have to use catkin_make_isolated and maybe this will solve the problem.

wxmerkt, 2020-03-13:
@parakhm95 `catkin build` does work - it's a ThirdParty released project with a `package.xml` so does work. The issue above appears to be from not running `rosdep install --from-paths ./ -iry` to install required dependencies.

wxmerkt, 2020-03-15:
Closing - please reopen if problem persists.

closed 2020-03-15 by wxmerkt

AndreV84, 2021-05-03:
won't it work with opencv 4.1?

auntkb, 2021-10-04:
as @AndreV84 said, still not working with opencv 4.1.

AndreV84, 2021-10-04:
However, in case of use of the following it will work with in-build opencv 3.2 that comes with apt package, in my opinion
```
apt install ros-melodic-apriltag-ros
```