AprilRobotics/apriltag_ros #68: Phantom tags detected

dean4ta, 2019-11-12:
Hi I have noticed that when running the continuous apriltag detector, tags that are not obviously present are being detected intermittently (~every 30s to 2mins)
This is the scene that the tag detector is analyzing:
[image]

here is the output on this scene:
'''
process[apriltag_ros_continuous_node-1]:  started with  pid   [14227]
[ INFO] [1573585127.6199681741: Initializing nodelet with 4 worker threads.
[ INFO] [1573585127.6406806711: Loaded tag config: 4, size: 0.347,  frame_name:  tag4
[ INFO] [1573585127.6407277061:  Loaded  tag  config: 118,  size: 0.482,  frame_name:  tag118
[ INFO] [1573585127.6408075941:  Loaded  tag  config: 114,  size: 0.482,  frame_name:  tag114
[ INFO] [1573585127.640859780]:  Loaded  tag  config: 119,  size: 0.482,  frame_name:  tag119
[ INFO] [1573585127.6409054071:  Loaded  tag  config: 117,  size: 0.482,  frame_name:  tag117
[ INFO] [1573585127.6409383671:  Loaded  tag  config: 113,  size: 0.482,  frame_name:  tag113
[ INFO] [1573585127.6409841911:  Loaded  tag  config: 100,  size: 0.482,  frame_name:  tag100
[ INFO] [1573585127.6410274771:  Loaded  tag  config: 102,  size: 0.482,  frame_name:  tag102
[ INFO] [1573565127.6410800201:  Loaded  tag  config: 103,  size: 0.482,  frame_name:  tag103
[ INFO] [1573585127.6411117341:  Loaded  tag  config: 108,  size: 0.482,  frame_name:  tag108
[ INFO] [1573585127.6411400311:  Loaded  tag  config: 109,  size: 0.482,  frame_name:  tag109
[ INFO] [1573585127.6411698901:  Loaded  tag  config: 111,  size: 0.482,  frame_name:  tag111
[ INFO] [1573585127.6411943851:  Loaded  tag  config: 101,  size: 0.482,  frame_name:  tag101
[ WARN] [1573585127.6422456651:  remove_duplicates  parameter  not  provided.  Defaulting  to  true
[ WARN] [1573585178.9513108811: Requested description of standalone tag ID [502], but no description was found...
[ WARN] [1573585249.5462589011: Requested description of standalone tag ID [500], but no description was found...
WRN:  Matrix  is  singular.
WRN:  Matrix  is  singular.
[ WARN] [1573585540.3505753751: Requested description of standalone tag ID [356], but no description was found...
WRN:  Matrix  is  singular.
WRN:  Matrix  is  singular.
[ WARN] [1573585745.5476507831: Requested description of standalone tag ID [46], but no description was found...
[ WARN] [1573585763.5498190051: Requested description of standalone tag ID [98], but no description was found...
[ WARN] [1573585897.3455831111: Requested description of standalone tag ID [554], but no description was found...
[ WARN] [1573586016.5486510421: Requested description of standalone tag ID [451], but no description was found...
[ WARN] [1573586101.3513115651: Requested description of standalone tag ID [354], but no description was found...
[ WARN] [1573586205.7525610501: Requested description of standalone tag ID [46], but no description was found...
[ WARN] [1573566233.3479171751: Requested description of standalone tag ID [544], but no description was found...
[ WARN] [1573586348.7532520831: Requested description of standalone tag ID [472], but no description was found...
[ WARN] [1573586467.3435749991: Requested description of standalone tag ID [46], but no description was found...
'''

Here is an example of a phantom tag detection's pose
'''
id: [100]
size: [0.482]
pose: 
  header: 
    seq: 13019
    stamp: 
      secs: 1565902950
      nsecs: 731362772
    frame_id: "pylon_camera_optical"
  pose: 
    pose: 
      position: 
        x: -0.00338844753806
        y: 0.0019613117084
        z: -0.00213273995678
      orientation: 
        x: 0.142591284097
        y: 0.690506810118
        z: -0.625784842158
        w: -0.333558693794
'''

It seems like the tags are detected due to noise in the image. Any suggestions on how to debug or resolve?

rgreid, 2019-11-13:
You have a pretty degenerate scene with the repeated square patterns filled by "noisy" gray.  The Apriltag papers discuss bit error rates; in particular Table I and II in [1] suggest you might want to try the 41h12 tag family. 

[1] M. Krogius, A. Haggenmiller, and E. Olson, “Flexible Tag Layouts for the AprilTag Fiducial System,”

wxmerkt, 2020-10-21:
Closing due to old age - please reopen if issue still persists

closed 2020-10-21 by wxmerkt