AprilRobotics/apriltag_ros #70: APril tag cannot Build nor find packages to launch

AhmetMericOzcan, 2019-12-11:
```
at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/asd/c_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/asd/c_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

I am using ubuntu 16.04 with Kinectic distrubition. I can build with catkin_make_isolated, but this time cannot find apriltag package when I try to run it. Even though I used source devel/setup.bash.

Anyone ?

wxmerkt, 2019-12-12:
Did you run `rosdep install` to get all dependencies? On Kinetic, you may have to clone the AprilTag repository into your source workspace as well.

wxmerkt, 2020-02-26:
Closing until updates / further details received

closed 2020-02-26 by wxmerkt