AprilRobotics/apriltag_ros #71: Unable to create a subscriber

viswanarayanans, 2020-01-03:
I have been trying to create a subscriber to /tag_detections topic, which is of type apriltags_ros::AprilTagDetectionArray. I have tried to follow the instructions as mentioned in the [other issue](https://github.com/AprilRobotics/apriltag_ros/issues/25), but I found out that apriltags2_ros package as apriltag_ros. I made the changes, I got a compilation error that "apriltag_ros/AprilTagDetectionArray.h" is not found. I checked the package, but still couldn't find this file. Can someone help me with this?

wxmerkt, 2020-01-05:
Hi, we need some more details to respond to your issue. Which system are you on and how did you install apriltag_ros?

danielCrawley, 2020-02-26:
I got it to work after doing the following in my ros package:

In CMakeLists.txt:
`find_package(catkin REQUIRED COMPONENTS 
   roscpp
   rospy
   std_msgs
   apriltag_ros # added this one
)`
Then in package.xml I added:
`<build_depend>apriltag_ros</build_depend>
<build_export_depend>apriltag_ros</build_export_depend>`

Lastly, in my cpp file I included it like this:
`#include <apriltag_ros/AprilTagDetection.h>
#include <apriltag_ros/AprilTagDetectionArray.h>`

After running catkin_make, there were no errors - I'll update you once I flesh it out some more and get it running.

wxmerkt, 2020-02-26:
Thank you for the details @danielCrawley - this is what it's expected. I will close the issue now until we receive further detail.

closed 2020-02-26 by wxmerkt