AprilRobotics/apriltag_ros #72: image_proc sync problem

m4xr1sk, 2020-02-20:
I'm using image_proc to rectify the image of my camera.
Without image_proc the tag detection works but obvioulsy the orientation is wrongly estimated.

With image_proc the tag detection doesn't work and now I'm getting this warning:

```
[ WARN] [1582182434.473958456]: [image_transport] Topics '/image_proc_fisheye1/image_rect' and '/image_proc_fisheye1/camera_info' do not appear to be synchronized. In the last 10s:
        Image messages received:      301
        CameraInfo messages received: 0
        Synchronized pairs:           0
```

How can I solve?
Is this the problem that tags are not detected any more?

parakhm95, 2020-03-13:
How are you publishing the camerainfo? Also, is your distortion matrix correct? Because from my experience, things don't work well for fisheye parameters with image_proc. It throws an error about distortion matrix not being available and then quits.

wxmerkt, 2020-10-21:
Closing due to old age - please reopen if issue still persists

closed 2020-10-21 by wxmerkt