AprilRobotics/apriltag_ros #73: Apriltag.c: failed to allocate hamming decode table. Reduce max hamming size

Soakedcardinal, 2020-03-25:
I got this error running the aptriltag_ros_continuous_node using family 48h12 (The one with empty space in the middle).

following https://github.com/AprilRobotics/apriltag/issues/68,

I changed apriltag_detector_add_family(td_, tf_) to apriltag_detector_add_family_bits(td_, tf_, 1) in common_functions.cpp and it works after recompiling. 

I don't know what's going on here but I figured this is an issue related to the implementation of newer tag families in apriltag 3.

wxmerkt, 2020-10-21:
How much memory does your machine have? I noticed that for `tagStandard41h12` the node in its standard form right now uses 384 MB and `tagCustom48h12` 4.7 GB. 

@mkrogius, do you recommend to set the bits to 1 depending on family or in general?

mkrogius, 2020-10-21:
This should only affect the larger tag families. The memory consumption is essentially O(n choose k) == O(n!/(k! * (n-k)!)) where n is the number of tags in the family and k is whatever you set the number of bits to (as long as k << n). So to reduce memory requirements you can either reduce the number of tags in the family or reduce the number of bits of correction.

Maybe we should add an easier way for people to say that they don't want to use all of the possible tags in a family? I imagine most people use only a few hundred tags instead of the thousands or tens of thousands of tags in the larger families.

Soakedcardinal, 2020-11-11:
@wxmerkt I'm running this on an Odroid XU4 with 2GB LPDDR3 RAM

wxmerkt, 2021-04-07:
There is now an option to set the maximum hamming distance thanks to #93 - this should address the problems raised in this issue.

closed 2021-04-07 by wxmerkt