AprilRobotics/apriltag_ros #78: Apriltags in Gazebo

HappySamuel, 2020-09-24:
Hi

Is there any tutorials on implementing apriltag into Gazebo?

Best,
Samuel

wxmerkt, 2020-10-21:
Hi Samuel,
If should work with (1) simulated cameras and (2) placing objects with the tags as textures in the environment. I am not aware of a tutorial describing both steps but they are likely out there.

Hope you've been able to resolve it.

Best,
Wolfgang

closed 2020-10-21 by wxmerkt

shonigmann, 2020-10-21:
Hi Samuel,

Its probably a bit late, but I very recently got Apriltag working in Gazebo, so if you have any questions I might be able to help.

Also, if it helps, I made a AR marker cube for gazebo and the corresponding tag config file, which you can find here: https://grabcad.com/library/april-tag-ar-marker-cube-1