AprilRobotics/apriltag_ros #79: Tags detected (in image and tf) but no data in /tag_detections

fnilius, 2020-09-29:
Hi,
I installed apriltag_ros using apt (sudo apt ros-melodic-apriltag-ros).
The detection seem to work fine (tags are well detected in the /tag_detections_image topic, and are correctly published to the /tf topic).
However, there seem to be no data published to the /tag_detections topic (which I would like to use to get the covariance of the pose).

Do you have an idea what might be the issue?

OS : ubuntu 18.04
Ros distribution : melodic
Configuration files : in the attached zip file (it also contains a sample of the output of the camera_info)
[config_files.zip](https://github.com/AprilRobotics/apriltag_ros/files/5299835/config_files.zip)

Thanks a lot in advance
Best
Felix NILIUS

PS : I also tried an installation from source, but didn't succeeded

umor3, 2020-10-08:
Hey,
did you manage do solve the problem? I'm stuck with the same problem. The detection is working but the position of the tag relative to the camera is (0,0,0)
I looked the code in common_functions.cpp and `pose.pose = tag_detection_array.detections[i].pose.pose.pose;` is (0,0,0) too.

umor3, 2020-10-08:
Hey fnilius,
I was having the same error but managed to solve it. Is your camera calibrated? When you look into rqt -> topic monitor -> your_camera/camera_info what do you get?

To calibrate your camera you need to follow this tutorial: http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration 
My printer (A4) produced 0.025 m squares. So my command was: 
`rosrun camera_calibration cameracalibrator.py --size 8x6 --square 0.025 image:=/your_camera/image_raw camera:=/your_camera`

After finishing the tutorial you get a yaml-file in `/tmp/calibration data.tar.gz` named `ost.yaml`
Depending on your camera-launch-file you have to reference this file. I'm using the usb_cam-package (http://wiki.ros.org/usb_cam) and I had to add the command 
`<param name="camera_info_url" type="string" value="file://$(find your_pkg_path)/launch/ost.yaml" />`
and you have to match the camera_name from the calibration file to the one referenced in your launch file:
`<param name="camera_name" value="camera_name_in_yaml"/>` e.g. `narrow_stereo` or edit the yaml-file to match your camera-launch-file name.

Hope this solves your problem too
Cheers Robert

wxmerkt, 2020-10-21:
Thanks for the clear tutorial on how to resolve it @umor3.

I'll close this issue now - @fnilius please reopen and add detail if still persists.

closed 2020-10-21 by wxmerkt

chrisywong, 2021-05-03:
I'm running into this issue too (tags detected in `/tag_detections_image` and in `tf`, but nothing is published to `/tag_detections`). My `camera_info` is being published and seen in `rqt_topic`. Any suggestions?