AprilRobotics/apriltag_ros #84: AR tag pose Off?

swiz23, 2020-12-29:
Not sure if I'm doing something wrong, but...

 I'm currently developing using the Intel RealSense SR305 camera, and decided to compare the pose output from AprilTag ROS with a pose estimated using the pointcloud from the SR305 camera. The output is below. Note that the grid size is set to 0.2 meter spacing.

[image]

As can visually be seen, the pointcloud approach says the tag is approximately 57 cm away from the camera. Using a measuring tape verifies that this is accurate. However, the tag_detections topic says that the tag is about 100 cm away from the camera!

The orientation seems to match pretty well though...
I'm using the camera/color/image_raw and the camera/color/camera_info topics outputted from the realsense2 ros node as input to the apriltag_ros node. Do you think the camera_info topic from the realsense camera just has a straight-up terrible calibration matrix? But if that's the case, why does the colored pointcloud look accurate?

Note that I've noticed that this issue happens with the realsense D435 camera as well. FYI - I'm using ROS Kinetic on Ubuntu 16.04 with the latest code.

rgreid, 2020-12-29:
> As can visually be seen, the pointcloud approach says the tag is approximately 57 cm away from the camera. Using a measuring tape verifies that this is accurate. However, the tag_detections topic says that the tag is about 100 cm away from the camera!

First step debugging this is to check whether the tag sizes are set correctly in [tags.yaml](https://github.com/AprilRobotics/apriltag_ros/blob/master/apriltag_ros/config/tags.yaml
). Does the size match the distance shown by the red arrow below? This image is taken from the [AprilTag User Guide.](https://github.com/AprilRobotics/apriltag/wiki/AprilTag-User-Guide)

[image]

swiz23, 2020-12-29:
Yep, that was the issue. I had assumed that the tag size was defined from the outer edges of the black pixels. After changing to the correct tag size, the pose estimation was much more accurate (as shown below).

[image]

While looking through the User Guide you linked, I noticed a section on the Coordinate system.

>The coordinate system has the origin at the camera center. The z-axis points from the camera center out the camera lens. The x-axis is to the right in the image taken by the camera, and y is down. The tag's coordinate frame is centered at the center of the tag, with x-axis to the right, y-axis down, and z-axis into the tag.

From your description, my camera frame seems correctly oriented. The AR tag though has its z-axis going out of the tag, but you say it should go into it?

Looking behind the AR tag, the frame looks like...

[image]

So the x-axis is to the right, the y-axis is down, but the z-axis points out of the tag?
Thanks so much for your help!

rgreid, 2020-12-30:
> So the x-axis is to the right, the y-axis is down, but the z-axis points out of the tag?

That's the correct behavior. See [here](https://github.com/AprilRobotics/apriltag_ros/blob/master/apriltag_ros/src/common_functions.cpp#L322) in the ROS wrapper source. 
```
    //   - camera frame: looking from behind the camera (like a
    //     photographer), x is right, y is down and z is straight
    //     ahead
    //   - tag frame: looking straight at the tag (oriented correctly),
    //     x is right, y is up and z is towards you (out of the tag).
```

closed 2020-12-30 by rgreid