AprilRobotics/apriltag_ros #89: No transformation from <..> to <..> error when continuous_detection.launch is executed.

canberkgurel, 2021-01-21:
Disclaimer: This is a question. So, label it accordingly if you have the privileges.,

I created a TF tree where I defined the pose of each apriltag w.r.t. the /map frame. I want to localize my camera by detecting the apriltags whose pose's are defined in the tf tree. 

e.g., my tf tree for tag0. 
```
<node pkg="tf2_ros" type="static_transform_publisher" name="april_tag_id_0" args="0.000 2.620 0.720 0 0 0 1 map tag0" />
```

I have also modified the apriltag_ros/config/tags.yaml, added the tag0 information.

```
standalone_tags:
  [
    {id: 0, size: 0.16, name: tag0}
  ]
```
I'm using rviz to visualize the tf structure. I start by running my tf tree. The coordinate frame appears in rviz, everything is fine so far. Then, I run the `continuous_detection.launch` file and suddenly rviz throws a warning.

> No transform from [camera] to frame [map]
> No transform from [tag0] to frame [map]

The transformation from tag0 to map was defined, it gets preempted when `continuous_detection.launch` starts running. Why do I need to define the transformation from `/map` to `/camera`? That's the unknown in my equation, it needs to be estimated  fusing `/map` >> `/tag0` >> `/camera` transformations.

Nosille, 2021-02-03:
Hope I can explain this well.  

when using tf2, transforms have a defined direction from parent->child, and each child can only have one parent.  You have defined map->tag0 and are expecting AprilTags to define the transform from tag0->camera, but AprilTags is actually defining a transform from camera->tag0 which conflicts your definition of tag0's parent.  It would be necessary to go into the Apriltag code and reverse the direction of the transform to achieve what you are attempting.

canberkgurel, 2021-02-06:
Hi @Nosille ,

I solved the problem by doing the transformation in my rospy node. I don't actually need the apriltag_ros to do the transformation for me, it's a nice to have yet not a must. I plan to look into this later in case I want to outsource this operation to the apriltag_ros.

> but AprilTags is actually defining a transform from camera->tag0

I did a quick test and confirmed your statement. If this is the case, then this vital information needs to be available in the wiki.ros documentation (http://wiki.ros.org/apriltag_ros). I read it carefully and didn't see it in there. If it's already there, please let me know. 

> when using tf2, transforms have a defined direction from parent->child

this doesn't make sense to me, if a map-to-tag transformation is defined then the tag-to-map is also defined -- which happens to be the inverse of the original transformation. So are you telling me that I need to define two separate static_transform_publishers one from map-to-tag and another from tag-to-map ?

Nosille, 2021-02-06:
Just like the publisher that you setup, Apriltags_ros relies on the tf2_ros package (http://wiki.ros.org/tf2) for transforms.  This directionality in the parent->child definition comes from that package.  I suggest reading its documentation for the how and why.  I think it mostly comes down to making it impossible to create loops in the tf tree.  If there are multiple paths between frames in the tf tree, which one is right?  This one parent, multiple children rule prevents that problem.

Apriltags_ros's authors simply assumed that the camera was already in the tf tree and the tag was not.  Therefore, it is coded to generate the camera->tag transform.  If you want it the other way, I suggest finding the line of code that calls the tf_publisher and reversing the transform.  It is located in the block of code starting at line 390 in common_functions.cpp.  You will need to both invert the transform, and swap camera_tf_frame_ and detection_names[i] in the call to send_transform.

rgreid, 2021-02-07:
Hi @canberkgurel. This is normal for ROS, as @Nosille suggested, the TF tree is a _directed tree_. The ROS REP explains a little [here](https://www.ros.org/reps/rep-0105.html#relationship-between-frames) and the TF2 docs explains [here](http://wiki.ros.org/tf2/Design#The_system_doesn.27t_need_to_know_about_the_configuration_before_hand_and_can_handle_reconfiguring_on_the_fly):  
>Value: Generic system for any configuration.
>Implementation: Use directed tree structure. It allows fast traversal(order n where n is the depth of the tree) when evaluating a transform. It can be reconfigured simply by redefining a link. It does not require any structure verification or maintenance of the data structure, except for maintaining a sorted linked list of data for each link.

Would you like to update the documentation here http://wiki.ros.org/apriltag_ros to clarify?

Feel free to reopen if you have other questions! Cheers.

closed 2021-02-07 by rgreid

shawkontzu, 2021-10-06:
I agree that the transformation /camera -> /apriltags can be a little confusing when we want to fuse the apriltag pose to robot_localization. Since tf only allows each frame to have a single parent frame, and a common tf tree is something similar to this

`/world -> /map -> /odom -> /base_link -> /camera_link -> /apriltags`

We have the ground truth /map -> /apriltags and we have the predicted transformation /camera_link -> /apriltags, but what we want is /map -> /base_link and fuse this pose information to robot_localization. One approach I could think of is since

`Tmap->apriltags = Tmap->base_link * Tbase_link->camera_link * Tcamera_link->apriltags`

so 

`Tmap->base_link = Tmap->apriltags * (Tcamera_link->apriltag)^-1 * (Tbase_link->camera_link)^-1`

We can name this calculated pose data to something like /landmark, and set a 0 transformation from /base_link -> /landmark, now tf tree knows that the new pose information /landmark has 0 transformation from /base_link, we just need to publish the calculated /landmark for fusing the apriltag data. This seems to be a workaround if we want to use apriltags for robot localization.