AprilRobotics/apriltag_ros #110: What is the unit of visual label size in config/tags.yaml

Zhang-Qinghui, 2021-11-30:
What is the unit of size？m of cm?
//////////////////////
standalone_tags:
  [
    {id: 10, size: 0.15},
    {id: 20, size: 0.1},
    {id: 30, size: 0.07}
  ]
tag_bundles:
  [
    {
      name: 'my_bundle',
      layout:
        [
          {id: 0, size: 0.05, x: 0.0000, y: 0.0000, z: 0.0, qw: 1.0, qx: 0.0, qy: 0.0, qz: 0.0},
          {id: 4, size: 0.05, x: 0.0548, y: -0.0522, z: 0.0, qw: 1.0, qx: 0.0, qy: 0.0, qz: 0.0},
          {id: 3, size: 0.05, x: -0.0580, y: -0.0553, z: 0.0, qw: 1.0, qx: 0.0, qy: 0.0, qz: 0.0},
          {id: 2, size: 0.05, x: 0.0543, y: 0.0603, z: 0.0, qw: 1.0, qx: 0.0, qy: 0.0, qz: 0.0},
          {id: 1, size: 0.05, x: -0.0582, y: 0.0573, z: 0.0, qw: 1.0, qx: 0.0, qy: 0.0, qz: 0.0}
        ]
    }
  ]

wxmerkt, 2021-11-30:
It's metres

Zhang-Qinghui, 2021-12-04:
Thanks for your reply. Your work is excellent！
I found that the obtained XY coordinate is very accurate, but the Z coordinate is very inaccurate, and the error of Z coordinate is more than 40 cm. Why does this happen? Do you have any suggestions

wxmerkt, 2021-12-05:
Intuitively, there could be two reasons: (1) not measuring the tag size correctly - different families are measured in different ways, consult the AprilTag paper/repository; (2) the intrinsics of the camera need calibration. If xy are accurate, my hunch would be to start looking at (1) first.

Zhang-Qinghui, 2021-12-06:
Thanks for your reply. Your suggestion is very effective.
 The tag I use is tag36h11. I now measure the size of the tag from the black border, rather than from the outermost white border. Now the error is very small, within 10cm. In addition, I will consult and confirm the tag size measurement method of tag36h11.

closed 2021-12-06 by wxmerkt