AprilRobotics/apriltag_ros #113: Ability to invert camera->tag transformation for each Standalone/ Bundle April Tag

JohnTGZ, 2022-01-07:
## The problem that this enhancement is targeting
When multiple cameras are used to detect a standalone/bundle April Tag, it leads to the april tag frame having multiple camera frames as parents.

## Example
Let's say we use April Tag with ID0 from the family "tag36h11", and we set the apriltag_ros_continuous_node param `publish_tf` to true, we will end up with the following tf being published `"<CAMERA1>_optical_frame" -> "tag_0"`, `"<CAMERA2>_optical_frame" -> "tag_0"` etc.
If we use multiple cameras, we ultimately end up with the frame "tag_0" having multiple parents.

## Other possible solutions
Of course, there are a few ways around this, one of which is to set `publish_tf` to false, take the desired bundle/standalone pose from `/tag_detections` message, publish an inverted frame transformation in the form of `"tag_0" -> "<CAMERA1>_optical_frame"` where 'tag_0' is now the parent instead of the child. 

## Proposal
To deal with the problems mentioned above, there could be an option to invert the published transformation between the camera and the april tag.
If you think such an enhancement would be useful for say, extrinsic camera calibration or frame manipulation. This is would be something Im happy to contribute to. Comments are appreciated too.

JSaunders97, 2022-02-21:
Hi @JohnTGZ ,

I've come across this same issue when doing extrinsic camera calibration, I think it would be really great if there was an option for this in apriltag ros! It seem like it would be relatively easy to implement as well.

Let me know if you decide to implement this?

JohnTGZ, 2022-02-22:
Yes I would definitely like to implement this, hopefully by mid-march