AprilRobotics/apriltag_ros #120: catkin build is failed

adamanov, 2022-03-07:
Hello, I am trying to build a workspace with `catkin build` command, however, it is failed. 
The following error is received 
```
lto1: fatal error: bytecode stream in file ‘apriltag_pywrap.o’ generated with LTO version 4.2 instead of the expected 6.2
compilation terminated.
```

or more general 

```
Errors     << apriltag:make /home/robolab/jackal_ws/logs/apriltag/build.make.001.log
lto1: fatal error: bytecode stream in file ‘apriltag_pywrap.o’ generated with LTO version 4.2 instead of the expected 6.2
compilation terminated.
lto-wrapper: fatal error: x86_64-linux-gnu-gcc returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[2]: *** [apriltag.cpython-36m-x86_64-linux-gnu.so] Error 1
make[1]: *** [CMakeFiles/apriltag_python.dir/all] Error 2
make: *** [all] Error 2
cd /home/robolab/jackal_ws/build/apriltag; catkin build --get-env apriltag | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << apriltag:make                            [ Exited with code 2 ]  
Failed    <<< apriltag                                 [ 0.4 seconds ]         
Abandoned <<< rslidar_sdk                              [ Unrelated job failed ]
Abandoned <<< apriltag_ros                             [ Unrelated job failed ]
Abandoned <<< apriltag_detector                        [ Unrelated job failed ]
Abandoned <<< omp_gripper_moveit_config                [ Unrelated job failed ]

```

I tried also `catkin_make_isolated` 
bu also received a similar error 
```
==> Processing plain cmake package: 'apriltag'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/robolab/jackal_ws/build_isolated/apriltag/devel'
==> make -j4 -l4 in '/home/robolab/jackal_ws/build_isolated/apriltag/devel'
[ 80%] Built target apriltag
[ 82%] Generating apriltag.cpython-36m-x86_64-linux-gnu.so
[ 87%] Built target apriltag_demo
[ 92%] Built target opencv_demo
lto1: fatal error: bytecode stream in file ‘apriltag_pywrap.o’ generated with LTO version 4.2 instead of the expected 6.2
compilation terminated.
lto-wrapper: fatal error: x86_64-linux-gnu-gcc returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
CMakeFiles/apriltag_python.dir/build.make:61: recipe for target 'apriltag.cpython-36m-x86_64-linux-gnu.so' failed
make[2]: *** [apriltag.cpython-36m-x86_64-linux-gnu.so] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/apriltag_python.dir/all' failed
make[1]: *** [CMakeFiles/apriltag_python.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'apriltag': 
  Command '['make', '-j4', '-l4']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/robolab/jackal_ws/build_isolated/apriltag && make -j4 -l4

Command failed, exiting.
```

wxmerkt, 2022-03-07:
Hi @adamanov 
Which Ubuntu/ROS version are you on? Python 3.6 sounds as if it's Ubuntu 18.04? You can try deactivating AprilTag's Python wrapper or installing AprilTag from binaries: `sudo apt install ros-$ROS_DISTRO-apriltag`

wxmerkt, 2022-03-10:
As long as the version in package.xml mirrors the one distributed via the buildfarm, no. We regularly make releases - check the difference in the repo (commits since last release)

closed 2022-03-10 by wxmerkt

adamanov, 2022-03-10:
> As long as the version in package.xml mirrors the one distributed via the buildfarm, no. We regularly make releases - check the difference in the repo (commits since last release)

Thank you very much for your help and answer :)

wxmerkt, 2022-03-10:
I've realised that the releases in the binaries missed quite a few recent PRs. I just made new releases which hopefully should be available in a few weeks. In the meantime, building from source gives you immediate access to the new features/bug fixes.