AprilRobotics/apriltag_ros #125: Initialize detections_ with empty struct

maximilianwulf, 2022-04-07:
Maybe a special case of usage here, but

Running the node and not providing any input messages gives the following error when closing the node

```
^C[apriltag_ros_continuous_node-2] killing on exit
apriltag_ros_continuous_node: /tmp/binarydeb/ros-noetic-apriltag-3.2.0/common/zarray.h:132: zarray_size: Assertion `za != NULL' failed.
```

This is because in the [constructor](https://github.com/AprilRobotics/apriltag_ros/blob/master/apriltag_ros/src/common_functions.cpp#L157) of the TagDetector 

```
   detections_ = NULL;
```

The [desctructor](https://github.com/AprilRobotics/apriltag_ros/blob/master/apriltag_ros/src/common_functions.cpp#L166) wants to destroy `detections_` with 

```
  apriltag_detections_destroy(detections_);
```

But `zarray_size` [asserts](https://github.com/AprilRobotics/apriltag/blob/master/common/zarray.h#L132) if the pointer points to NULL.

Is it possible to initialize `detections_` with something else than `NULL` ?

wxmerkt, 2022-04-07:
Hi @maximilianwulf,
Thank you very much for reporting this issue - I guess we haven't observed this since we rarely build in Debug mode. A simple check if `detections_` is a valid pointer, i.e. `if (detections_) apriltag_detections_destroy(detections_);` might fix it?

maximilianwulf, 2022-04-07:
Hi @wxmerkt, yes that should solve. Do you want me to test and submit that issue?

wxmerkt, 2022-04-07:
Yes, can you please test this and make a PR? Many thanks! :-)

maximilianwulf, 2022-04-08:
https://github.com/AprilRobotics/apriltag_ros/pull/126

PR #126 (2022-04-08): Add guard before destroy method in TagDetector destructor
Fixes #125 

Was tested on our setup.
closed 2022-04-08 by wxmerkt