AprilRobotics/apriltag_ros #127: what is the purpose of zarray_get(detections_, i, &detection) and how set  H Matrix

Wuxinxiaoshifu, 2022-04-29:
there is an H matrix  in addImagePoints(detection, standaloneTagImagePoints) , i'm wondering how H is get  int your function?

some details:
1.homography_project(detection->H, tag_x[i], tag_y[i], &im_x, &im_y);
2.
// assuming that the projection matrix is:
// [ fx 0  cx 0 ]
// [  0 fy cy 0 ]
// [  0  0  1 0 ]
//
// And that the homography is equal to the projection matrix times the model matrix,
// recover the model matrix (which is returned). Note that the third column of the model
// matrix is missing in the expresison below, reflecting the fact that the homography assumes
// all points are at z=0 (i.e., planar) and that the element of z is thus omitted.
// (3x1 instead of 4x1).
//
// [ fx 0  cx 0 ] [ R00  R01  TX ]    [ H00 H01 H02 ]
// [  0 fy cy 0 ] [ R10  R11  TY ] =  [ H10 H11 H12 ]
// [  0  0  1 0 ] [ R20  R21  TZ ] =  [ H20 H21 H22 ]
//                [  0    0    1 ]
//
// fx*R00 + cx*R20 = H00   (note, H only known up to scale; some additional adjustments required; see code.)
// fx*R01 + cx*R21 = H01
// fx*TX  + cx*TZ  = H02
// fy*R10 + cy*R20 = H10
// fy*R11 + cy*R21 = H11
// fy*TY  + cy*TZ  = H12
// R20 = H20
// R21 = H21
// TZ  = H22

Wuxinxiaoshifu, 2022-04-29:
i also wondering whether the H is known or unknown  before running the homography_project function?

Wuxinxiaoshifu, 2022-04-29:
in others words, the function  of apriltag_detector_detect calculate the H?

wxmerkt, 2022-11-21:
Has your question been resolved? I am not sure I am following based on the pointers given so far

closed 2022-11-30 by wxmerkt