AprilRobotics/apriltag_ros #130: pass back corner detections

inventor2525, 2022-06-04:
Just wondering if there's interest in [this](https://github.com/HorseRobotics/apriltag_ros/commit/f1088855d3c91b1c2d130b92510d8b77fa1380b4). Possibly made optional.

Reasons we wanted the change:
1) To try removing rotational ambiguity in limited instances using extra points, possibly inner corners
2) Add other adjacent markers in the future (color calibration).
3) Track tag looking for blur resistant extra markers (possibly until [ArucoDCF](https://github.com/Ric92/aruco_dcf) [not the current ros node]).

julled, 2022-12-27:
I would be also in receiving the corner and center detections as e.g. Detection2D messages

christianrauch, 2022-12-27:
> I would also interested in receiving the corner and center detections as e.g. Detection2D messages

The corners are in the [`Point[4] corners`](https://github.com/christianrauch/apriltag_msgs/blob/0c42545ff53a1a2aa98ff5fe045e3001a316e884/msg/AprilTagDetection.msg#L7). Is this sufficient?

The `vision_msgs/Detection2D` only contains a `BoundingBox2D` which does not really match a tag viewed at an angle (i.e. skewed).

christian-rauch, 2022-12-27:
> I would also interested in receiving the corner and center detections as e.g. Detection2D messages

The corners are in the [`Point[4] corners`](https://github.com/christianrauch/apriltag_msgs/blob/0c42545ff53a1a2aa98ff5fe045e3001a316e884/msg/AprilTagDetection.msg#L7). Is this sufficient?

The `vision_msgs/Detection2D` only contains a `BoundingBox2D` which does not really match a tag viewed at an angle (i.e. skewed).