AprilRobotics/apriltag_ros #132: ROS2 april-tag

trentatsa, 2022-07-21:
If there is a plan to release an apriltag-ros package for ROS2, will that be anytime soon ?

wxmerkt, 2022-07-21:
Hi @trentatsa - please take a look at #76 where there's a link to https://github.com/christianrauch/apriltag_ros

closed 2022-07-21 by wxmerkt

trentatsa, 2022-07-21:
Hi @wxmerkt ,unfortunately that package doesn't have all the features such as bundle tags etc etc.  Will there be no official release by AprilRobotics ?

wxmerkt, 2022-07-21:
I'll let @christian-rauch comment on the ROS2 side of things. In either case, support would likely have to be provided by the community - happy to look at any PR to add this functionality

LSinterbotix, 2022-08-24:
The port in #114 works as an almost drop-in replacement for my organization's purposes. For context, we primarily use the single image detector node and the single image tag detection service.

trentatsa, 2022-08-26:
Hi @LSinterbotix and @wxmerkt, thanks for the suggestion. @wep21 great work on the port. So I have tested out the port, seem like single tag server/ client works perfectly however the continuous detector node has a bug. It detects the tag, publishes the detection but then the node dies. Maybe @wep21 can check this ?

wep21, 2022-08-27:
@trentatsa Could you tell me how to reproduce the node dying? I will check that.

trentatsa, 2022-08-29:
Hi @wep21, no problem, appreciate the help. 
Running ROS Galactic on Jetpack 5.1 on Jetson Xavier which is based on Ubuntu 20.04. Going to give it a bash on generic PC with base Ubuntu 20.04 on it. 
All parameters are default as found in your repository. 
Let me know if you need any other info!

wep21, 2022-08-29:
@trentatsa I'm sorry, but I can't reproduce the issue. I tested with the following commands.
```
ros2 launch apriltag_ros continuous_detection.launch.xml
```

```
ros2 run image_publisher image_publisher_node /home/daisuke/Downloads/apriltagrobots_overlay.jpg --ros-args -r image_raw:=image_rect -r __ns:=/camera_rect
```

I added some modifications to remove conflicts with `master` branch.
Could you pull them and try it with the commands above? If there are any errors when the node die, could you share the error messages?

trentatsa, 2022-09-01:
@wep21 so I have tested it and it seems that its a platform thing (Jetson Xavier Nx). Works perfectly on a normal PC. I think it might be an opencv thing as Jetson dont use generic Ubuntu opencv. Will do some digging.

orjano-max, 2023-02-14:
@trentatsa Did you find any solution to this compatibility issue on the Jetson Xavier? I am currently trying to run it on a Jetson AGX Xavier running Jetpack 5.0.2 and ROS2 Foxy. 
Experiencing a similar issue where i run: 

```bash
 ros2 launch apriltag_ros tag_realsense.launch.py
```
And once the tag appears in the image screen the node crashes and i get this error:

[component_container-1]   what():  OpenCV(4.5.4) /home/ubuntu/build_opencv/opencv/modules/core/src/matrix.cpp:250: error: (-215:Assertion failed) s >= 0 in function 'setSize'
[component_container-1] 
[ERROR] [component_container-1]: process has died [pid 34486, exit code -6, cmd '/opt/ros/foxy/lib/rclcpp_components/component_container --ros-args -r __node:=tag_container -r __ns:=/apriltag']

wxmerkt, 2023-02-14:
Are you using @christian-rauch's apriltag_ros for ros2? If so it might be better to post in the corresponding repository

trentatsa, 2023-02-14:
@orjano-max, I did find a fix.

Comment out  the following lines in continuous_detector.cpp

// Publish the camera image overlaid by outlines of the detected tags and
  // their payload values
  //if (draw_tag_detections_image_) {
   // tag_detector_->drawDetections(cv_image_);
   // tag_detections_image_publisher_.publish(cv_image_->toImageMsg());
  //}

orjano-max, 2023-02-14:
@wxmerkt I will, thanks for the heads up.
@trentatsa Thank you for the response, i realized i did not use the same version of the apriltag package.
Fixed the issue by building **vision_opencv** from source, more info: https://github.com/christianrauch/apriltag_ros/issues/9#issuecomment-1429778790