AprilRobotics/apriltag_ros #133: Inaccurate pose

viveksood97, 2022-07-29:
I have been trying to get pose of an april tag cube in gazebo but the pose estimation seems to be very inaccurate. As my robot moves towards the tag, the tag seems to move closer(move in the axis of motion) to the robot. Is this a known issue?

wxmerkt, 2022-11-21:
Hi Vivek,
It's difficult to provide insights - it might be due to your fixed frame (visualisation) or how you set things up. Generally, using AprilTags works quite well once the configs are set up.

closed 2022-11-21 by wxmerkt

nevoltom, 2024-09-05:
Hi, I am having the same issue. I have a simulation in gazebo where I have a tag36h11 of size 1x1m. In config I set the size to 0.8. When I am with the camera directly above the tag, I get translation in x and y in meters, which is a nonsense. Only somehow accurate estimate is the height above the tag. Do you know what could be the issue?