AprilRobotics/apriltag_ros #147: load_parameters: unable to set parameters [roslaunch apriltag_ros continious_detection.launch]

cakarmustactalone, 2023-04-07:
Hi guyz, 

When I run the launch file via `roslaunch apriltag_ros continious_detection.launch`  I get this error:

> load_parameters: unable to set parameters (last param was [/apriltag_ros_continuous_node/queue_size=1]): cannot marshal None unless allow_none is enabled
> Traceback (most recent call last):
>   File "/opt/ros/noetic/lib/python3.10/site-packages/roslaunch/__init__.py", line 347, in main
>     p.start()
>   File "/opt/ros/noetic/lib/python3.10/site-packages/roslaunch/parent.py", line 316, in start
>     self.runner.launch()
>   File "/opt/ros/noetic/lib/python3.10/site-packages/roslaunch/launch.py", line 676, in launch
>     self._setup()
>   File "/opt/ros/noetic/lib/python3.10/site-packages/roslaunch/launch.py", line 663, in _setup
>     self._load_parameters()
>   File "/opt/ros/noetic/lib/python3.10/site-packages/roslaunch/launch.py", line 354, in _load_parameters
>     r  = param_server_multi()
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 892, in __call__
>     return MultiCallIterator(self.__server.system.multicall(marshalled_list))
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 1122, in __call__
>     return self.__send(self.__name, args)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 1461, in __request
>     request = dumps(params, methodname, encoding=self.__encoding,
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 981, in dumps
>     data = m.dumps(params)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 514, in dumps
>     dump(v, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 536, in __dump
>     f(self, value, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 589, in dump_array
>     dump(v, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 536, in __dump
>     f(self, value, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 607, in dump_struct
>     dump(v, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 536, in __dump
>     f(self, value, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 589, in dump_array
>     dump(v, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 536, in __dump
>     f(self, value, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 589, in dump_array
>     dump(v, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 536, in __dump
>     f(self, value, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 607, in dump_struct
>     dump(v, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 536, in __dump
>     f(self, value, write)
>   File "/usr/lib/python3.10/xmlrpc/client.py", line 540, in dump_nil
>     raise TypeError("cannot marshal None unless allow_none is enabled")
> TypeError: cannot marshal None unless allow_none is enabled

I did not change anything in continious_detection.launch file, it still exist. 

Here is my launch file:

```
<launch>
  <!-- set to value="gdbserver localhost:10000" for remote debugging -->
  <arg name="launch_prefix" default="" />
  <!-- configure camera input -->
  <arg name="camera_name" default="/camera_rect" />
  <arg name="image_topic" default="image_rect" />
  <arg name="queue_size" default="1" />
  <!-- apriltag_ros continuous detection node -->
  <node pkg="apriltag_ros" type="apriltag_ros_continuous_node"
  name="apriltag_ros_continuous_node" clear_params="true"
  output="screen" launch-prefix="$(arg launch_prefix)">
    <!-- Remap topics from those used in code to those on the ROS network -->
    <remap from="image_rect"
    to="$(arg camera_name)/$(arg image_topic)" />
    <remap from="camera_info"
    to="$(arg camera_name)/camera_info" />
    <param name="publish_tag_detections_image" type="bool"
    value="true" />
    <!-- default: false -->
    <param name="queue_size" type="int"
    value="$(arg queue_size)" />
    <!-- load parameters (incl. tag family, tags, etc.) -->
    <rosparam command="load"
    file="$(find apriltag_ros)/config/settings.yaml" />
    <rosparam command="load"
    file="$(find apriltag_ros)/config/tags.yaml" />
  </node>
</launch>
```
Any idea what is going on? 

Thanks.
Best  Regards.

wxmerkt, 2023-04-07:
Hi @cakarmustactalone,
Could you share more detail on your setup? I don't think Python 3.10 is supported by ROS Noetic, so this seems odd/non-standard

cakarmustactalone, 2023-04-07:
@wxmerkt  I 'm in arch linux(ros-noetic) which is supported by ROS but experimental. I have builded via catkin_make_isolated. Which python version you have?

wxmerkt, 2023-04-07:
Could you try removing the launch-prefix argument from the launch file? I wonder if the empty default string is treated as a None.

cakarmustactalone, 2023-04-07:
@wxmerkt here is what I get only removing launch_prefix argument:
 ```
RLException: [/home/jaguar/Projects/ros/catkin_ws/src/apriltag_ros/launch/continuous_detection.launch] requires the 'launch_prefix' arg to be set
The traceback for the exception was written to the log file
```
when I removed launch-prefix from <node-pkg= "apriltag_ros" ... as shown below

```
  <node pkg="apriltag_ros" type="apriltag_ros_continuous_node"
  name="apriltag_ros_continuous_node" clear_params="true"
  output="screen" >
```

Still got same error.

Regards.

SmileJayjhx, 2023-05-08:
met the same problem today, the problem is wrong indentation format in tags.yaml.
You can give it a try with nothing in `standalone_tags:[]` and launch the continuous_detection.launch, if that works, you should check your own settings.