AprilRobotics/apriltag_ros #148: Failed to load nodelet [/apriltag_ros_continuous_node] of type [apriltag_ros/ContinuousDetector] even after refreshing the cache: Could not find library corresponding to plugin apriltag_ros/ContinuousDetector

HowardQian201, 2023-04-12:
When following the apriltag_ros tutorial, I've run into an error where the library corresponding to plugin apriltag_ros/ContinuousDetector could not be found. I've attached screenshots of both the terminal output and the nodelet_plugins.xml file. I have not altered the apriltag_ros package in any way. Any help on this matter would be greatly appreciated!

[image]
[image]

wxmerkt, 2023-04-12:
Have you tried opening a new terminal, sourcing the workspace, and re-running the above command? We cannot help diagnosing this much further - it works well for most users so might be a discovery or local issue

HowardQian201, 2023-04-12:
Thanks for the comment - I've tried a few things including what you mentioned. I'm trying to understand what the paths are referring to in the nodelet_plugins.xml file. For example, what file/directory does "lib/libapriltag_ros_continuous_detector" refer to? Thanks again