AprilRobotics/apriltag_ros #157: No tag is detected! possibly due to [WARN] remove_duplicates parameter not provided. Defaulting to true

Karin-Sugi, 2023-11-30:
Hi,
I am a beginner of apriltag. Now, i tried to use it with ros and followed its tutorial.

After setting all yaml files, I ran `roslaunch apriltags2_ros continuous_detection.launch` to get no image in `/tag_detections_image` topic. (I had no idea to check detection success with other method)

I'm sure my camera works fine & topic name of camera info is correct. How can i fix it??
The warning message follows.
```
process[apriltag_ros_continuous_node-1]: started with pid [6990]
[ INFO] [1701337254.855863409]: Initializing nodelet with 4 worker threads.
[ INFO] [1701337254.910183256]: Loaded tag config: 0, size: 0.06, frame_name: tag_0
[ INFO] [1701337254.910217094]: Loaded tag config: 1, size: 0.06, frame_name: tag_1
[ INFO] [1701337254.910259886]: Loaded tag config: 2, size: 0.06, frame_name: tag_2
[ INFO] [1701337254.910270820]: Loaded tag config: 3, size: 0.06, frame_name: tag_3
[ INFO] [1701337254.910276905]: Loaded tag config: 4, size: 0.06, frame_name: tag_4
[ INFO] [1701337254.910283172]: Loaded tag config: 5, size: 0.06, frame_name: tag_5
[ WARN] [1701337254.910676213]: remove_duplicates parameter not provided. Defaulting to true
```

Also, my two yaml files in `/opt/ros/melodic/share/apriltag_ros/config` are below.

```
# in settings.yaml
tag_family:        'tag36h11' # options: tagStandard52h13, tagStandard41h12, tag36h11, tag25h9, tag16h5, tagCusto    m48h12, tagCircle21h7, tagCircle49h12
tag_border:        1          # added
tag_threads:       2          # default: 2
tag_decimate:      1.0        # default: 1.0
tag_blur:          0.0        # default: 0.0
tag_refine_edges:  1          # default: 1
tag_refine_decode: 0          # added
tag_refine_pose:   0          # added
tag_debug:         0          # default: 0
#max_hamming_dist:  3          # default: 2 (Tunable parameter with 2 being a good choice - values >=3 consume la    rge amounts of memory. Choose the largest value possible.)
# Other parameters 
publish_tf:        true       # default: false
#transport_hint:    "raw"      # default: raw, see http://wiki.ros.org/image_transport#Known_Transport_Packages f    or options 
```

```
# in tags.yaml
standalone_tags:
  [
      {id: 0, size: 0.06},
      {id: 1, size: 0.06},
      {id: 2, size: 0.06},
      {id: 3, size: 0.06},
      {id: 4, size: 0.06},
      {id: 5, size: 0.06},
  ]
tag_bundles:
 [
 ]
```

yl-wang996, 2023-12-20:
@Karin-Sugi Could you explain why you closed this issue, I have the same problem currently. May I ask, did you resolve this question and how? thanks in advance.

peci1, 2024-05-21:
The warning you reported has nothing to do with the filter working wrong.

Just a question, don't you run apriltag_ros in a virtual machine on ARM MAC computers? We've noticed ARM MACs run the detector with no detections, but also with no failure.